NUbots/NUbots #7: Per Robot Config

deevus, 2015-04-17:


deevus, 2015-04-17:
@TrentHouliston 

I remember you saying that config files are aggregated in some way? There's so much code in the project so it would be great if you could point me in the general direction.

Can you explain how it currently works in explicit terms? What command do I call to generate a config? Without any changes what is the current process? (once again I'd like an explicit explanation so I can investigate the invocation)

Sorry if I'm being a pain here but the sheer amount of code is pretty overwhelming.

comicxmz001, 2015-04-17:
Same question to Simon. The explicit terms.

comicxmz001, 2015-04-17:
@deevus 
If you want the config files about servos, sensors, limbIDs, they are located at NUbots\shared\messages\input .

Some other config files, such as field configs, you may find it from NUbots\modules\support\configuration

That's what I know, hope it helps. :P

deevus, 2015-04-17:
@comicxmz001 Thanks. I'll have a look

deevus, 2015-04-17:
Starting point: https://github.com/NUbots/NUbots/blob/develop/tools/cmake/NUClearModuleBuilder.cmake#L40-L60

Thanks @mitchellmetcalfe

deevus, 2015-04-17:
https://github.com/NUbots/NUbots/blob/develop/tools/scripts/send.py
https://github.com/NUbots/NUbots/blob/develop/modules/support/configuration/ConfigSystem/src/ConfigSystem.h
https://github.com/NUbots/NUbots/blob/develop/shared/messages/support/Configuration.h

TrentHouliston, 2017-07-27:
Fixed

closed 2017-07-27 by TrentHouliston