NUbots/NUbots #12: Unhandled std::exception in LinuxCamera

Bidski, 2015-06-29:
Running robotlocalisation role (and probably other vision related roles) generates the following error (after some amount of time)
```
ERROR: Identifier:  Read Camera
ERROR: Identifier:  std::tuple<NUClear::dsl::Trigger<NUClear::dsl::Every<33333333, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > >, NUClear::dsl::Options<NUClear::dsl::Single> >
ERROR: Identifier:  modules::input::LinuxCamera::LinuxCamera(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > const&)#1}
ERROR: Unhandled Exception:  std::exception
ERROR: Identifier:  
ERROR: Identifier:  std::tuple<NUClear::dsl::Trigger<messages::input::Image>, NUClear::dsl::Options<NUClear::dsl::Single, NUClear::dsl::Priority<(NUClear::EPriority)2> > >
ERROR: Identifier:  modules::support::NUbugger::provideVision()::{lambda(messages::input::Image const&)#1}
ERROR: Unhandled Exception:  std::exception
```

Bidski, 2015-06-29:
Tested with a camera with two microphones, similar error occurs.
```
src/central_freelist.cc:322] tcmalloc: allocation failed 16384 
ERROR: Identifier:  Classify Image
ERROR: Identifier:  std::tuple<NUClear::dsl::Trigger<NUClear::dsl::Raw<messages::input::Image> >, NUClear::dsl::With<messages::vision::LookUpTable>, NUClear::dsl::With<NUClear::dsl::Raw<messages::input::Sensors> >, NUClear::dsl::Options<NUClear::dsl::Single> >
ERROR: Identifier:  modules::vision::LUTClassifier::LUTClassifier(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda(std::shared_ptr<messages::input::Image const> const&, messages::vision::LookUpTable const&, std::shared_ptr<messages::input::Sensors const> const&)#4}
ERROR: Unhandled Exception:  std::exception
ERROR: Identifier:  Read Camera
ERROR: Identifier:  std::tuple<NUClear::dsl::Trigger<NUClear::dsl::Every<33333333, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > >, NUClear::dsl::Options<NUClear::dsl::Single> >
ERROR: Identifier:  modules::input::LinuxCamera::LinuxCamera(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > const&)#1}
ERROR: Unhandled Exception:  std::exception
ERROR: Identifier:  Read Camera
ERROR: Identifier:  std::tuple<NUClear::dsl::Trigger<NUClear::dsl::Every<33333333, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > >, NUClear::dsl::Options<NUClear::dsl::Single> >
ERROR: Identifier:  modules::input::LinuxCamera::LinuxCamera(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda(std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long long, std::ratio<1ll, 1000000000ll> > > const&)#1}
ERROR: Unhandled Exception:  std::exception
```

Bidski, 2015-06-30:
This could potentially be a memory exhaustion issue that is potentially related to the ball detector seeing a ball.

TrentHouliston, 2017-07-27:
Outdated issue

closed 2017-07-27 by TrentHouliston