NUbots/NUbots #15: Getup does not trigger if robot is not lying completely flat on the ground

mmetcalfe, 2015-07-06:
Sometimes we have to nudge the robot to trigger the getup.
The getup getup should trigger if the robot's orientation is far from vertical for a long enough period of time (with low variance).

josiahw, 2015-07-06:
This is a good solution. "long enough" is probably < 1s after the fall
because you have lots of sensor readings coming in.

The only thing to watch out for is that the walk positions the body quite
far from vertical (I had trouble with this last year). It would be helpful
to take the walk's offset from vertical into account when trying to
determine what "horizontal" means for the robot (this probably means having
a different sensitivity to fallen in the forward and backward directions).

ysims, 2022-08-26:
Get up check works well.

closed 2022-08-26 by ysims