NUbots/NUbots #17: CM730 board disconnects then reconnects as ttyUSB1

mmetcalfe, 2015-07-06:
By default, the CM730 appears as ttyUSB0, however upon disconnecting, it tends to reconnect as ttyUSB1.
If we could detect this occurrence and have the robot reconnect to the CM730 on ttyUSB1, we would reduce the frequency at which we would have to service the robots during RoboCup games.

josiahw, 2015-07-06:
If it disconnects, it might be worth starting at usb0 and scanning up to
say 5 until you hit it. The usb1 reconnect happens because there hasn't
been a timeout on the usb0 socket at the point where the board is
re-detected. I have similar issues using arduinos sometimes.

Bidski, 2015-07-06:
we could possibly set up a udev rule to always symlink the CM730 USB device to a specific "file" in /dev. In theory, it should then not matter whether it is on ttyUSB0 or ttyUSB1 (or some other ttyUSBX dev) as we should always be able to refer to it with our own custom symlink

TrentHouliston, 2017-07-27:
Fixed using udev rules

closed 2017-07-27 by TrentHouliston