NUbots/NUbots #384: Codebase builds are not caching properly.

KipHamiltons, 2021-03-15:
When you run `./b build`, something is not caching properly, requiring 150-250 files to be recompiled (even if nothing has been changed).

Steps to replicate:
1. build the codebase with `./b build`
2. run `./b build` again, without making changes. It will go on to rebuild around 150-250 files, no matter what.

This behaviour was introduced by #342 (Changing all header files to `.hpp`). Building on master at the commit before caches properly, and building after does not.

TrentHouliston, 2021-03-15:
My guess for this issue is that somewhere a file is being generated but it's not listed as the file that's generated in CMake

e.g. CMake thinks it's running a script that makes `blah.h` but it actually makes `blah.hpp` and since everyone includes `blah.hpp` it builds fine. But next time CMake tries to build it notices `blah.h` doesn't exist so runs the script again, so `blah.hpp` gets a modified timestamp starting the whole build again.

KipHamiltons, 2021-03-15:
Fixed by #385

closed 2021-03-15 by KipHamiltons