NUbots/NUbots #397: Buildkite agents should have their cache cleared regularly

KipHamiltons, 2021-03-24:
Currently, the buildkite agents don't have any of their docker caches cleared ever. This could end up taking up all of their harddrive space eventually. It would be nice if they had a regular systemd (probably weekly) task run which cleared each of their caches.
Setting this up would involve writing a script/systemd task to do it, putting that script/task on the existing buildkite machines, then updating the [instructions for setting up buildkite agents in NUbook](https://nubook.nubots.net/system/foundations/ci-system#install-and-configure-buildkite).

Here's an example list of things to purge/bash commands to do it:
```sh
docker buildx prune -af
docker container prune -f
docker image prune -af
docker volume prune -f
docker system prune -af
# possibly unnecessary, since I think this doesn't build up now that we use buildx...
docker builder prune -f
```

Y0sh1dk, 2021-03-24:
I can sort that out soon :+1: ,
It probably doesnt need to be systemd and just a cron job is fine, is that bash script something I would put in [here](https://github.com/NUbots/NUbots/tree/master/doc)?

KipHamiltons, 2021-03-24:
This is probably an issue (as are most of the ones I am writing up) which we might want to wait to do until after webots/robocup things. 
If your existing PRs are waiting on other people, and you don't have any defined work to do on the webots stuff, then first port of call is probably to bug people for things to do for our webots efforts. If that fails, then we should look to start picking up more projects like this one. This one is probably pretty simple to knock over though. :shrug: 

Sidenote - @Bidski was keen on using a systemd task to do it, instead of a cron job. personally, I am indifferent :p

Bidski, 2021-03-24:
I am also indifferent, but systemd feels more right to me. I am also wondering if we could set it up to monitor hard drive space and then purging when X% of hard drive is occupied, which cant be done as a cron job (as far as i am aware)

KipHamiltons, 2021-03-24:
Interesting idea. One problem with that though is if we have a buildkite agent set up on a machine which has that amount of hard drive space taken up by other stuff (e.g. Josephus' lab computer, which might be filled with windows bloat :smirk: ), it could try to run it all the time. Although I suppose you might be able to set up a cooldown for it?

Y0sh1dk, 2021-03-24:
Instead of monitoring hard drive space directly, we could monitor image, container, volume and cache size with `docker system df`.

We also might run into a issue if it tries to clear caches as a build is in progress?

Bidski, 2021-03-24:
You can also set a different percentage on those machines. Or, in the case ofna persons lab computer, that person can handle it manually when they feel it is necessaary

Bidski, 2021-03-24:
Hard drive space would be more indicative than the cache size alone.

We should able to check for active containers to make sure there are no active builds. Failing that we can always make a lock file that prevents the clean up from running

JosephusPaye, 2021-03-24:
Hey! My computer in the lab is Mint (that now barely works).

Bidski, 2021-03-24:
Also, the purge should only affect dangling/unused images/containers, not ones that are currently in use

ysims, 2022-07-23:
GitHub Actions used now instead of Buildkite

closed 2022-07-23 by ysims