NUbots/NUbots #398: We should handle unused modules better

KipHamiltons, 2021-03-24:
Unused modules can become annoying if they're not in roles. Not being in roles means they aren't ever compiled, so things change under them making them more difficult to use if we ever want to unearth them.
Possible solutions include:
- A script to find unused modules. This could be a standalone solution, but that would require it to be ran manually, and deliberated on manually etc. This could also be used as part of other solutions. PRs (#48 #290) have been made to do this.
- A role which has all modules in it. this role could be optional to build, perhaps with a CMAKE option (analagous to -DBUILD_TESTS=ON), but we would get the buildkite agents to run it to make sure that nothing is broken unknowingly by things that aren't in "real" roles.  This role should be ignored by git, and would ideally be generated somehow.
- A b script module integrating those two ideas. It would generate the role, and optionally list the unused modules. the buildkite agents would use it to generate the role for their checks. This is the cleanest solution of those listed.

Something that these solutions don't cover though are modules which aren't connected to other modules. They might be in roles, but they might not emit or consume any messages which are used by the other modules in the roles they're in. This isn't a big issue, but it's something that could theoretically be covered by work on top of a graph of message interactions, such as that provided by #343.

KipHamiltons, 2021-05-17:
The first bullet point has been addressed (#290 has been merged).
The third bullet point should be the next action for this issue.

KipHamiltons, 2021-07-14:
As of right now, the following modules are unused, as found by `./b unused`:

| Module | Status |
| ---------- | -------- |
| motion::IKKick | Apparently going to be rewritten (# ) |
| support::VirtualCamera | Useful |
| behaviour::skills::GoalSaver | Dead |
| localisation::OdometryLocalisation | Dead |
| output::eSpeak | Dead |
| support::ReactionTimer | Dead |
| behaviour::tools::SpeakTest | Dead |
| behaviour::skills::WalkEngineStand | I don't know |
| behaviour::skills::WalkPathFollower | I don't know |

I am going to submit PRs to remove the ones I claim are dead.
@ysims @TheCoolKidsTable @Bidski: `WalkEngineStand`/`WalkPathFollower` - are they going to be used again ever?

ysims, 2021-07-14:
`WalkEngineStand`: I'm not sure how it'd perform with the QuinticWalk. We should probably use it.
`WalkPathFollower`: No idea, it looks like an old walk path planner. We're rewriting the walk path planner anyway and we don't use this one so 🤷 probably don't need it. Unless it actually does something else, but it doesn't look like it.

KipHamiltons, 2021-07-15:
I just pushed a branch `hamiltons/check-unused-modules` which generates a role which contains those modules which aren't in roles, then builds that role in the buildkite pipeline. This would make it impossible to pass buildkite checks with changes which break the unused stuff. I think it should address the problems presented in this issue.

All that's left to do on it is make it compile because the unused modules are very broken lmao.