NUbots/NUbots #430: We should make sure all our quaternions in odometry are unit quaternions

KipHamiltons, 2021-04-23:
When we swapped from arma to eigen, we stopped using our custom quaternion class, `UnitQuaternion`, in favour of eigen's quaternion class. This made us lose the unit norm guarantees afforded by that class.

It's worth going through the code (odometry - `MotionModel`, `SensorFilter`, and particle filters in particular) to make sure that we don't have any non-unit-norm quaternions being used anywhere. The easiest way to guarantee this might end up being making a unit quaternion class to wrap eigen's quaternion class, then only using that...

ysims, 2022-05-29:
@abihall

ysims, 2026-09-20:
@sebastianbrooker2 
This might still be worth doing, there's not many quaternions anymore but the handleful that exist might benefit from a wrapper. 
https://github.com/search?q=repo%3ANUbots%2FNUbots+Eigen%3A%3AQuaternion&type=code