NUbots/NUbots #445: Docker and DevOps Preparation for RoboCup 2021

KipHamiltons, 2021-04-29:
We have several things to prepare our codebase to play in the simulation setup at RoboCup 2021. Relevant (draft) links are the [server specification](https://cdn.robocup.org/hl/wp/2021/04/v-hsc_server_specification_draft1.pdf), which details the docker container requirements and setup for the robot containers, and the [API specs](https://cdn.robocup.org/hl/wp/2021/04/v-hsc_simulator_api_draft2.pdf). These were fetched from the [humanoid league RoboCup Forum](https://hl.forum.robocup.org/c/virtual-hsc/humanoid-soccer-competition/27) (and should be kept up to date in this thread, as they are finalised!)

The gist is that we need a robocup docker image, which contains our compiled codebase, ready to exchange gameplay information with a webots sim. The container requires:
- our compiled binaries
- our runtime dependencies/libraries
- GPU support
- extensive logging (we have 10 GB per robot per match, including the image(?)) in the folder `/robocup-logs`
- ability to read env variables and set config file values based on them

After preliminary discussions within the DevOps team (so literally me and Yoshi had a chat :joy: ), we are thinking we'll need at a minimum the following (**Plan of Attack**):
1. New build target i.e. something like `./b target robocup`, which will use the compiler flags relevant for the host system
2. Adapting `./b install` so that it can save a (probably compressed) set of binaries somewhere (probably in `NUbots/docker`, which is where we'll pull them into the robocup container from)
3. A multistage Dockerfile setup (probably), so that we can cache the overlapping layers from a normal `generic` image
4. Make a script (python or shell is fine) which initialises the players - as in, reads the environment variables, sets the configs based on them, then runs the robocup (or whatever role actually plays the game) role's binary
5. Set up automatic logging to files in the container's file system
6. (Probably) wrap up all of the functions required to make a robocup container in a new `./b` tool. Atm we are thinking about making an option/flag for the new tool to push the container to dockerhub. Either that, or another other tool to do that, so that it's definitely explicit and deliberate
7. Create a `config.json` which meets the specifications required for it
8. Make sure that we can network from the robocup container to some other arbitrary IP (basically we should dry run the whole setup, as soon as is feasible)

Some misc points:
- 2mins after containers start before game starts
 - 2mins for container to shutdown after game
 - Each robot runs on its own VM
 - Container run on host network (docker run --network host)
 - 10GB free storage at `/` (not persistent, for temp files)
 - 4 core vCPU
 - NVIDIA T4 GPU
 - We should perhaps (unironically) test things with fewer robots than 4. The network bandwidth is very jolly small.

KipHamiltons, 2021-04-29:
A preliminary look at the control flow of the proposed `./b robocup` tool which creates the robocup image is as follows:
1. `./b target robocup` # new target
2. `./b configure --clean` # probably optional, but on by default
3. `./b configure -- ... -DROLE_robocup=ON` # other roles off. Other flags included
4. `./b build`
5. `./b install --output-to-file=docker/robocup_files.tar` # or something
6. Some command to run the stage of the Dockerfile which pulls in the binaries and sets up the final environment
7. (if push to docker hub option selected) push the result to dockerhub

CMurtagh-LGTM, 2021-04-29:
I don't think the name `robocup` is good for the target or the tool, maybe something like `sim-robocup` or `webots-robocup` so we know that these things are for.

KipHamiltons, 2021-04-29:
> I don't think the name `robocup` is good for the target or the tool, maybe something like `sim-robocup` or `webots-robocup` so we know that these things are for.

Yea, something with `webots` in it is probably a good idea. I agree with you that calling it robocup is probably a bad idea on second thought lol

Y0sh1dk, 2021-05-18:
We are currently building and running images using a new `b` tool called `webots` :partying_face: :partying_face: 

`./b install` has been modifed to be capable of outputing binaries and runtime dependencies to a local directory, which are then used to build a docker image from. A entrypoint script in the image currently reads all required environemnt args (doesnt set them in config files yet), and then runs the specified role.

What still needs to be done:
 - Wait for comms to merge into master, or rebase onto comms branch and test out networking from within the container
 - Set config files with values read from environment args
 - Test out image on AWS
 - Setup logging to `/robocup-logs`
 - New build target with optimisations for AWS hardware
 - GPU support?
 - Different roles for different types of players? (goalie, striker etc)

KipHamiltons, 2021-05-18:
tremendous! sounds like a draft PR is in order!

Y0sh1dk, 2021-05-20:
Its worth noting that the `ROBOCUP_SIMULATOR_ADDR` should always be used as the address to connect to the Webots controller as the mapping `(teamcolour, player_number) -> server port` may not always be consistent.

JosephusPaye, 2021-07-01:
Looks like we did most of the things listed here. This issue could probably be closed.

closed 2021-07-01 by KipHamiltons