NUbots/NUbots #495: GameController Responses in Webots

ysims, 2021-06-01:
Currently, when running the official RoboCup controller our robot experiences many penalties until it gets sent into the time out corner. We need to track down the problems, which are probably related to the way we handle GameController messages.

LachlanCourt, 2021-06-01:
I might be able to help with this? What's the best way to start?

ysims, 2021-06-01:
Make a role that is identical to robocup but with webots instead of darwin::hardwareio.

Then run the robocup controller in webots and the role in NUbots. Config for the robocup controller is in the nubots/webots fork.

ysims, 2021-06-01:
Now I'm wondering if it does have issues or not. I might be misremembering and it's just the keyboardwalk that has this problem (obvs because it doesn't do game controller things). Still should test and verify it though.

ysims, 2021-06-07:
The issue has been fixed with a change in configuration file `team_1.json` over at NUbots/webots. We also respond the GameController as of #503.

closed 2021-06-07 by ysims