NUbots/NUbots #587: We should audit our branches

KipHamiltons, 2021-06-24:
We have 140 "active" branches. These should be checked for things worth merging and pruned if they're truly stale.

ysims, 2026-09-20:
Currently 286 branches.
@sebastianbrooker2 this is an issue that will never really be resolved. There are some tasks that need to be done semi-regularly - perhaps once or twice a year. These include PR and branch auditing, issue auditing, and code (module, utility) auditing. NUbook also has some auditing tasks, such as broken links and outdated information. An issue isn't really the best method of making sure this happens, since it gets pushed down, however if you're changing your method of project management perhaps you can have a 'regularly tasks' board or similar for this types of issues.

sebastianbrooker2, 2026-09-20:
yeah I should create a regular tasks board. thanks for the suggestion