NUbots/NUbots #588: Requested Changes to Head Behaviour

abihall, 2021-06-24:
As extension to Head Behaviour: #581 

Current observations and recommended changes:
- There are some floating TODO's in the code, they need to be tracked down and the required changes made
- The HeadBehaviourSoccer.cpp doesn't utilize the localisation balls, it relies heavily on vision balls rather than using both, might be a good change to do both
- Commented code seems to be functional, but #is not being used and unsure of why its commented out
- Commenting is a must for the whole file
- Some decent logs with good descriptions of status would be nice as well

KipHamiltons, 2021-06-24:
Further than that, the requested changes indicate the `HeadBehaviourSoccer` is in need of a rethink.

AlanaNoonan, 2021-06-24:
In config file, possible changes:
- Max head pitch is quite large, the robot would self collide before this. A test would be good to determine the angle value for which the head collides with the torso, or where it can see it's torso.
- There is a comment on line 28, that the pitch limit is a function of yaw which needs to be calculated properly. This should be resolved, or the comment should be removed.

LachlanCourt, 2021-06-26:
Time associated with then sensor_measurements coming from webots gets out of sync with NUClear clock. Should this be a new issue or is it only related to this head behaviour problem?

KipHamiltons, 2021-06-27:
that sounds like an everything problem. I would probably make a new issue @LachlanCourt

KipHamiltons, 2021-07-14:
This should be actioned. @weepeefa

ysims, 2023-04-20:
Issue is now out-of-date - head behaviour has had multiple refactors since this issue. See #1021 for newer issue on head behaviour.

closed 2023-04-20 by ysims