NUbots/NUbots #610: ROS Bridge

KipHamiltons, 2021-06-29:
ROS has many libraries and tools. It would be cool if we could use them. Making some middleware for communicating with a ROS bridge so that we can use them would sweet.
Until that's done, if there are any ROS specific libraries or tools which we want to use and they have permissive licences, we can look at refactoring into a module or shared library directly.

ysims, 2026-09-20:
@ZeoNyph you made a ROS bridge for K1 vision, right? Although whether to have one at all is an open question. I think the overhead was undesirable iirc.

ZeoNyph, 2026-09-20:
Yep! Theoretically it can be done via some form of IPC (I went with shared memory), but it was a hassle to setup for what I had to do. It wasn't so bad for NUbridge since it was a one-way bridge exclusively for ROS2 Image and Pose messages being sent by Booster's ROS2 firmware. The problem came from having to make sure both ends had the same message header, which again is fine for one message but when juggling a lot it can be harder to manage.

Given enough time and effort, maybe setup can be done to create middleware that reads from a single source of truth (ROS2 .msg to .proto converter?) and make it easier to have a bridge.

With regards to overhead, there is none to begin with for the K1 as it already uses ROS2 under the hood for its motion and perception daemons, so we wouldn't be introducing more by integrating ROS2.