NUbots/NUbots #611: Plot Juggler could be interesting for NUsight charts

KipHamiltons, 2021-06-29:
@TheCoolKidsTable has mentioned that [Plot Juggler](https://github.com/facontidavide/PlotJuggler) is worth investigating. Apparently it has features that chart view does not. Please elaborate @TheCoolKidsTable

KipHamiltons, 2021-06-29:
github magic :D

TheCoolKidsTable, 2021-06-29:
It looks to be fairly straightforward to setup. I think it'll handle the large amount of data coming in better than charts. Has some nice features like multiple charts at once, plotting data on both x and y (useful for stuff like position). Simple drag and drop data. 

Downside is it will have to sit outside standalone from  nusight I'm pretty sure.

TrentHouliston, 2021-06-29:
Seems pretty cool!, you should be able to just write a simple module to grab the NUsight charts and others and use one of the many many streaming interfaces it supports!

KipHamiltons, 2022-08-02:
Did we not add this?

JosephusPaye, 2022-08-02:
The PR to add it (#878) is not merged yet. We'll close this once that gets in.

JosephusPaye, 2022-10-08:
Fixed in https://github.com/NUbots/NUbots/pull/878

closed 2022-10-08 by JosephusPaye