NUbots/NUbots #613: NUbook monorepo

KipHamiltons, 2021-06-29:
I would like to discuss moving the NUbook source into this repo.
Some pluses:
- Much easier to update documentation at the same time as updating source
- Fewer repo's to manage/track
- API documentation generation would be local to the website it generates for. This would be much simpler than porting across generated HTML regularly, or other naive solutions I have thought of

Some minuses:
- Netlify has limited free builds - although we could probably work it out so that netlify only builds master. In that case we would have to work out some way to get the preview links for the doc pages

Please feel free to post your opinions in here

KipHamiltons, 2021-06-29:
I will also add that adding the NUsight source into this repo seems to have been a great success (at least to me)

ysims, 2021-06-29:
I would say another negative, which I assume is an issue for NUsight but I don't do any dev on that, is it's harder to find PRs. I can easily see all NUbook PRs on the NUbook repo. If they were interspersed with heaps of NUbots and NUsight PRs it would be harder to see them. But labels might help.

I think not having preview links is a big problem.

Another point is that NUbook does not only detail NUsight and robot code, it covers much more than that. While NUsight only interacts with robot code.

KipHamiltons, 2021-06-29:
> I would say another negative, which I assume is an issue for NUsight but I don't do any dev on that, is it's harder to find PRs. I can easily see all NUbook PRs on the NUbook repo. If they were interspersed with heaps of NUbots and NUsight PRs it would be harder to see them. But labels might help.

Labels are an excellent point. I think they largely mitigate issues with finding PRs of a particular type.

> I think not having preview links is a big problem.

I think we could basically do previews by generating the website locally, now that I think about it. You can browse the HTML in your browser by just putting the path to the file in the omnibox, then navigating it as normal.

> Another point is that NUbook does not only detail NUsight and robot code, it covers much more than that. While NUsight only interacts with robot code.

This is an interesting point. Could you please elaborate on why this would be bad? I would have thought that most of the stuff in the `doc` directory would fit in this category already, e.g. team qualification papers.

Bidski, 2021-06-29:
Labels would help to mitigate the difficulty of finding PRs, even for PRs to nubots code (e.g. localisation related PRs).

It might even be possible to spawn CI related jobs based on labels? Im not sure how github webhooks work, but im pretty sure buildkite can rejects jobs in certain circumstances? Cant netlify do the same and does that count as a build?

ysims, 2021-06-29:
> > I think not having preview links is a big problem.
> 
> I think we could basically do previews by generating the website locally, now that I think about it. You can browse the HTML in your browser by just putting the path to the file in the omnibox, then navigating it as normal.
>

Gatsby still needs to generate the pages. It makes it harder to get people to review the pages, particularly ex team members. @Bidski's idea of label based jobs would fix this if it's a thing.
 
> This is an interesting point. Could you please elaborate on why this would be bad? I would have thought that most of the stuff in the `doc` directory would fit in this category already, e.g. team [...]

I'm not a big fan of the `doc` directory and barely touch it. It feels very messy to have something like NUbook in NUbots/NUbots. NUbook is not completely dependent on NUbots/NUbots. Having docs for NUpbr in a completely separate repo seems strange to me, even if it is built to a website that is standalone. Also feels like its easier to lose track of NUbook/NUsight because they're buried in NUbots/NUbots, rather than seeing them immediately in the pinned repos.

Bidski, 2021-06-29:
Looks like the labels thing isnt a thing 😞 I think we would have to add things like `[no-docs]` or something to the PR title in order to get the info into netlify to filter out the job.

Still, labels are something we should be using anyway

BrendanAnnable, 2021-06-29:
Big fan of monorepos, they simplify a lot. I think its also nice to be able to update docs at the same time you update code. CI is the main complication though. Something I have meaning to do is avoid triggering C++ builds for NUsight-only related changes. e.g. check the directories touched by the commits, and if the changes are only within `/nusight2`, only build/deploy NUsight. We could something similar for NUbook.

ysims, 2021-06-29:
> Big fan of monorepos, they simplify a lot. I think its also nice to be able to update docs at the same time you update code. CI is the main complication though. Something I have meaning to do is avoid triggering C++ builds for NUsight-only related changes. e.g. check the directories touched by the commits, and if the changes are only within `/nusight2`, only build/deploy NUsight. We could do something similar for NUbook.
> 
> https://buildkite.com/docs/pipelines/conditionals

I'd be open to it if we can get this working and add in a NUbook doc check for PRs, something along the lines of someone that isn't the author has to tick a box saying NUbook has been appropriately updated. Not all PRs require NUbook changes so I don't think it should require a change in that folder... Or at least if we do that, allow it to be human overridable.

Bidski, 2021-06-29:
The box being ticked is what approving the PR means

ysims, 2021-06-29:
> The box being ticked is what approving the PR means

Need something to remind people though. I'm not going to remember unless I'm actively writing NUbook pages at the time. Some automatic thing that says 'go update NUbook'.

KipHamiltons, 2021-06-29:
We could make a github bot which posts a reply when people make a PR reminding them to update docs :laughing:

ysims, 2021-06-29:
> We could make a github bot which posts a reply when people make a PR reminding them to update docs 😆

That's an option. But you can ignore the bot - it's harder mentally to tick a box if the person hasn't actually done the thing (or at least I think it is...). Also bot can get lost in a long PR.

Bidski, 2021-06-29:
I think you can create PR templates that are filled with questions (tensorflow does it for their PRs). This might force people into putting more details into their PR descriptions

ysims, 2021-06-29:
> I think you can create PR templates that are filled with questions (tensorflow does it for their PRs). This might force people into putting more details into their PR descriptions

This is a good idea. GitHub often complaints to me about us not having a PR template when I go to our settings 😆

JosephusPaye, 2021-07-01:
Some responses to a few points raised:

> I think we could basically do previews by generating the website locally, now that I think about it.

Building the website locally is not the same as Netlify's deploy previews. The deploy previews are for viewing the site in a production environment, and they're more convenient for reviews, as Ysobel noted.

> Triggering NUbook CI based on a specific label:

Label based job triggers are not a thing as @Bidski mentioned. Also, NUbook's build and deploy CI is not handled by Buildkite, it's all done by Netlify. That said, it's possible to configure Netlify to [only build when there are changes to a specific folder](https://docs.netlify.com/configure-builds/common-configurations/ignore-builds/). We'll have to configure Buildkite to ignore changes to only those folders, like @BrendanAnnable is suggesting for C++ changes and NUsight builds.

---

Overall a monorepo will probably be more convenient, even though we'll have to find ways to deal with everything (issues, PRs, etc) being crowded in one place. Labels could help with this, though we'll have to remember to use them consistently.

KipHamiltons, 2021-07-08:
I wouldn't be against monorepo-ing everything if having docs for another repo is an issue. 
Personally I don't think it matters too much :p

Y0sh1dk, 2021-09-07:
I think we should consider this once https://github.com/NUbots/NUbots/pull/353 is merged in.

Deploying only on changes to the NUbook directory should be relatively easy on github actions with https://github.com/marketplace/actions/netlify-actions. In the NUbots workflow we can `paths-ignore` the NUbook directory, and in the NUbook workflow we can use `paths` to only run the corresponding workflow. See [here](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#onpushpull_requestpaths).

Reguarding the preview links, netlify-actions can be automatically configured to comment the preview link on PRs once the preview is deployed.