NUbots/NUbots #623: The Intel C++ Compiler

KipHamiltons, 2021-07-14:
The intel C compiler has highly regarded code generation (looking online, I found several stackoverflow threads and other endorsements, such as [this one](http://aosabook.org/en/llvm.html) (ctrl + f ICC to find it)). The licence is a pain though.

Someone should go to the effort of making an ICC target and then seeing if it performs radically differently or can generate more warnings or other benefits. This would require benchmarking of performance somehow (see #622)

ysims, 2026-09-20:
I do not know if this is relevant or not anymore @sebastianbrooker2, someone might want to dig a little to check.

sebastianbrooker2, 2026-09-20:
Sounds interesting, but compilation differences is beyond me. I'd need someone like @JohanneMontano to check this out. I'd only assume changing the way it compiles would take a bit of time, might be better in the long term but there'll be some weirdness or bugs that may appear? Not too sure how it works