NUbots/NUbots #650: Git-Aware Formatting

KipHamiltons, 2021-07-22:
I think it could be worth making a formatting flag which only formats the dirty files, or maybe the files committed in the last N commits. I don't see why this wouldn't be possible using git diffs and logs. This would make formatting faster, even for the typescript users.

JosephusPaye, 2021-07-24:
[lint-staged](https://www.npmjs.com/package/lint-staged) is a node package on that can be used to run prettier only on staged files. See https://stackoverflow.com/a/47610566

TrentHouliston, 2021-07-24:
It's easy enough to just update the `./b format` scripts to do it, the one that's running in CI already only runs on files that are committed to the git repository

CMurtagh-LGTM, 2021-09-07:
> It's easy enough to just update the `./b format` scripts to do it, the one that's running in CI already only runs on files that are committed to the git repository

Yes the CI runs on only files tracked by git, but I think Kip want's to format even less files.
The CI uses `git ls-files` which lists all tracked files, but I think `git diff --name-only main` would be better as it does not list files that haven't been changed from our known correctly formatted files.

KipHamiltons, 2021-09-07:
Precisely. That would be perfect.

PR #696 (2021-09-07): Format only changed files
closes #650
closed 2021-09-07 by CMurtagh-LGTM