NUbots/NUbots #651: Flag for configuring only a set of roles

KipHamiltons, 2021-07-22:
If you want to build only one role atm it's a bit painful, because you have to go through the full list in `./b configure -i` spamming space bar. Selecting a subset of the roles and configuring using those only is fairly straightforward (as demonstrated [here](https://github.com/NUbots/NUbots/blob/d89c7a2e16b6005a1663e4ac70c063de7a732966/tools/webots.py#L65) by an impressively neat solution from @JosephusPaye). Making this a proper thing for `./b configure` would be nice.

Happyman31, 2022-07-02:
Looking into maybe doing this. I will investigate how `./b configure` works and get some possible lists of set roles needed. From the example demonstrated, it should be somewhat easy to make a menu to select groups of roles to build with? More investigation needed!

KipHamiltons, 2022-07-02:
Happy to chat about details at some point. Hit me up on slack

PR #1031 (2023-08-16): Toggle Groups of Roles
Fixes #651

Adds flags to `configure` to set and unset groups of roles.
closed 2023-08-16 by JesseWilliamson