NUbots/NUbots #653: Make NUsights packet filter be able to deal with message id fields

TrentHouliston, 2021-07-23:
The current way that NUsight's message filter works is it ensures the previous packet is processed before sending a new one. The can cause problems for things like cameras where two packets of the same type arrive at the same time and because the left camera is read first only the left camera is ever updated.

The node.js server side of NUsight should be able to read the `message.id` field and create a separate filter for each message type, for each subtype

BrendanAnnable, 2021-07-23:
Note that not all messages have an `id` field, but it should be used if it exists. i.e. making the field name `id`, special.

e.g. [Image::id](https://github.com/NUbots/NUbots/blob/master/shared/message/input/Image.proto#L52)

For performance reasons we will also need to make sure that we only decode the `id` field from the packet, not the entire proto.

JosephusPaye, 2021-09-06:
I'm working on this at 4Tel, will port here when done.

closed 2021-10-10 by JosephusPaye