NUbots/NUbots #657: Autocomplete

CMurtagh-LGTM, 2021-07-29:
We can add [argcomplete](https://github.com/kislyuk/argcomplete) to the b script to add auto completion.

KipHamiltons, 2021-07-29:
@Y0sh1dk and I actually did look at this a while ago, then didn't follow through because it would require editing the `.bashrc`. I am ok with reevaluating our stance on that?

CMurtagh-LGTM, 2021-07-30:
At https://github.com/NUbots/NUbots/blob/master/nuclear/b.py#L112 we could add a line like
```py
module.register(subcommand.add_parser(components[-1]))
argcomplete.autocomplete(parser)
module.run(**vars(command.parse_args()))
```
which will not run the module when we want completions.

I think most people will be fine with putting things in their `.bashrc` as long as we explain what is going on.

CMurtagh-LGTM, 2021-07-30:
Also if you stop docker and run `./b build -h` it runs fine, so docker shouldn't be starting before we run `command.parse_args()`. 

Running `./b build -h` runs almost instantaneously so we should be fine for performance.

closed 2022-01-15 by CMurtagh-LGTM