NUbots/NUbots #689: Considering Podman as an alternative to Docker

CMurtagh-LGTM, 2021-09-03:
Reading articles and whatnot, there's some alternatives to docker that we should consider as it is no longer ubiquitous. For example kubernates has moved to interfacing directly with CRI-O.

Podman an OCI program with the same cli commands as docker. It has a bunch of security features that docker does not, e.g. no demon running as root, rootless containers that have the same permissions as the user that starts them. Companies, such as redhat, have moved away from docker to podman due to both security concerns and concerns in the projects direction https://www.linkedin.com/pulse/part-ii-why-docker-openshift-4-rhel-8-scott-mccarty.

I've got a branch with two minor changes that allow us to run most `b` commands with podman in rootless mode by doing `alias docker=podman` https://github.com/NUbots/NUbots/tree/cmurtagh/podman-compat. Also a cheeky `chmod -R o+w log` is needed for the user in the container to be able to write log files.

`b` commands that won't work in rootless mode without major changes
- `format`, due to user namespacing causing the current user to be mapped to root 
- `target --build`, due to buildx not being a command on podman. Pulling from dockerhub is still functional.

Should we migrate now? Probably not. Should we keep this as an option on the table for the future and see where the industry goes? Yes.

Can I please get you thoughts on this. @NUbots/devops

LachlanCourt, 2021-09-03:
I haven't read this material yet because I have a lot of uni on Fridays so I may not understand this properly and I'll read up later. But first pass of this, my main concern is that for things like robocup they expected us to submit a Docker image. It may be a number of years before all constituents catch up to newer practices and if we move too soon then it may cause issues that could easily be avoided by sticking with the status quo for now. Unless podman images are equivalent to docker images?

CMurtagh-LGTM, 2021-09-03:
> I haven't read this material yet because I have a lot of uni on Fridays so I may not understand this properly and I'll read up later. But first pass of this, my main concern is that for things like [...]

Podman images are equivalent to docker images.

Technically they aren't "docker images" but images defined by the (OCI)[https://opencontainers.org/] image [spec](https://github.com/opencontainers/image-spec)

Bidski, 2021-09-03:
Does podman provide better image/layer caching so that images properly contain their own cache?

CMurtagh-LGTM, 2021-09-03:
> Does podman provide better image/layer caching so that images properly contain their own cache?

Honestly I don't know exactly what you mean.
From the docs of buildah there is an option called layers that looks promising.
```
--layers bool-value

Cache intermediate images during the build process (Default is false).

Note: You can also override the default value of layers by setting the BUILDAH_LAYERS environment variable. export BUILDAH_LAYERS=true
```

Bidski, 2021-09-03:
> > Does podman provide better image/layer caching so that images properly contain their own cache?
> 
> Honestly I don't know exactly what you mean.
> From the docs of buildah/podman there is an option called layers that looks promising.
> ```
> --layers bool-value
> 
> Cache intermediate images during the build process (Default is true).
> 
> Note: You can also override the default value of layers by setting the BUILDAH_LAYERS environment variable. export BUILDAH_LAYERS=true
> ```

We had an issue with docker where the layer cache was only maintained on the local system and when you pushed the built image to dockerhub and pulled on another machine you would end up rebuilding the entire image even though the docker build context was unchanged.

This issue pushed us to use buildkit and buildx to get the layer cache to be stored in the built image, but it never properly fixed the issue.

If podman properly supports this it would be a good reason to migrate and would allow us to remove a lot of the recently added complexity in the `./b target` script

TrentHouliston, 2021-09-03:
If you used the GitHub actions it properly did the caching, the big problem was that build isn't used in the docker compose used by buildkite

JosephusPaye, 2021-09-03:
I think @wongjoel has used podman for a while now at CSIRO and could have some thoughts to share.

Bidski, 2021-09-03:
> If you used the GitHub actions it properly did the caching, the big problem was that build isn't used in the docker compose used by buildkite

Buildkite and docker-compose are being told to build with the [inline cache](https://github.com/NUbots/NUbots/blob/main/.buildkite/NUbots/docker-compose.yml#L10), apparently that still doesn't work though

CMurtagh-LGTM, 2021-09-03:
> Buildkite and docker-compose are being told to build with the [inline cache](https://github.com/NUbots/NUbots/blob/main/.buildkite/NUbots/docker-compose.yml#L10), apparently that still doesn't work [...]

It looks like [img](https://github.com/genuinetools/img) is an alternate buildkit builder. https://github.com/genuinetools/img/pull/279.

It also looks like inline caching is not yet implemented in podman.
```
   --cache-from
       Images to utilize as potential cache sources. Podman does not currently support caching so this is a NOOP. (This option is not available with the remote Podman client)
```

Bidski, 2021-09-03:
Thats disappointing. That would make this proposal a hard no from me. Docker at least supports caching, just storing the cache inline is a little broken

CMurtagh-LGTM, 2021-09-03:
> Thats disappointing. That would make this proposal a hard no from me. Docker at least supports caching, just storing the cache inline is a little broken

From what I've read podman seems to be more focused on running containers than building them. I think considering tools focused on building images from a dockerfile may be useful, e.g. [img](https://github.com/genuinetools/img).

wongjoel, 2021-09-03:
I've been using Singularity at CSIRO, though that's because it's a HPC specific container runtime. I actually still use Docker to make the images, then do a conversion to Singularity at the last step, because Singularity doesn't do caching, leading to super long build times (though it does produce an image as a file on the filesystem, so you can just move that file around instead of rebuilding if you need it on multiple computers).

Never used podman, but have been noting that Docker is facing some financial troubles

CMurtagh-LGTM, 2021-09-03:
So out build command has `--cache-to=type=inline,mode=max` but [buildkit](https://github.com/moby/buildkit#export-cache) says `inline` and `max` are mutually exclusive :thinking:

JosephusPaye, 2021-09-03:
> Never used podman, but have been noting that Docker is facing some financial troubles

Ah so it was Singularity, I somehow mixed that up with podman.

JosephusPaye, 2021-09-06:
This article has some tips on migrating, including some of the Docker features that are not available in Podman: https://marcusnoble.co.uk/2021-09-01-migrating-from-docker-to-podman/

KipHamiltons, 2021-09-15:
Without any build caching [(yet)](https://docs.podman.io/en/latest/markdown/podman-build.1.html#cache-from), I am going to close this. Feel free to reopen if you want to continue the discussion or make a fresh issue when they support caching.

closed 2021-09-15 by KipHamiltons