NUbots/NUbots #694: Documentation Generation - Next Steps

KipHamiltons, 2021-09-07:
1. The process of generating the docs with doxygen itself should be documented. [Probably here](https://nubook.nubots.net/guides/general/documentation#page-content) or somewhere near that.
2. Using doxygen, as in #676, doesn't support protobuf. We should figure out a way to generate protobuf docs and integrate them into the rest of the docs files.
3. #343 should be revived and integrated into the generated docs.
4. In-repo markdown files such as module readmes should be integrated into the generated docs.
5. Diagrams, should they be developed, should be integrated into generated docs.
6. The generated docs should be integrated into NUbook (see #613 for some related discussion). A monorepo is the easiest solution in many respects, but it will require some juggling with netlify builds, because of their limited free-build hours. I conjecture that github actions will make netlify build management easier though (#353).
7. The generated docs should be spruced up. They're not especially pretty atm, with the default theme, but doxygen has good facilities to improve the html it generates.

The most important thing to make the generated docs useful is going through our code and documenting it properly with appropriate [directive tags](https://nubook.nubots.net/guides/general/documentation#directives). This will require a sustained, concerted effort. It's very important, but it will never be urgent.

CMurtagh-LGTM, 2022-07-01:
- [ ] Somehow list the contents of each role? https://github.com/NUbots/NUbots/pull/862#pullrequestreview-1024303789

KipHamiltons, 2022-07-02:
Does #867 fulfil the requirements of the "spruce" item satisfactorily?

ysims, 2022-07-02:
> Does #867 fulfil the requirements of the "spruce" item satisfactorily?

I would say so