NUbots/NUbots #705: Goal::MeasurementType only ever having the value of CENTRE and RobotModel.hpp returning Zero matrix

abihall, 2021-09-10:
So, following talk on issue #450 and #315, it is found that the Goal::MeasurementType only ever takes the value of CENTRE and never that of LEFT_NORMAL or RIGHT_NORMAL. 

As a result, half of RobotModel.hpp will not be reached and most of the functions defined will return the Zero matrix since Goal::MeasurementType doesn't take on any value but CENTRE.

abihall, 2021-09-10:
In RobotModel.hpp, both getSquarePostCamSpaceNormal and getCylindricalPostCamSpaceNormal will return the Zero Matrix always. I can't see anywhere where these are used anyway. What is the point of having them there?

KipHamiltons, 2021-09-11:
We don't use those functions, nor do we have plans to use them. Their invalidity for the only measurement type we actually use compounds things to the point where they're properly dead and should definitely be removed.

Given that we don't have a consensus on the `MeasurementType` enum and only use one measurement type in the code, I think it should be removed too. We should update where it's used so it's clear exactly what the measurement means too, because without the enum, the extra context is valuable.

If you made PRs to remove those, I would approve them :stuck_out_tongue_winking_eye: (you should link #450 to the PR which removes the enum, so it closes once it's merged)

closed 2021-09-15 by abihall