NUbots/NUbots #736: The FileWatcher destructor can segfault

KipHamiltons, 2021-10-03:
I have been butting heads with a segfault and I have tracked down where the segfault originates. I can't replicate the issue on my machine - it only happens during a particular test.

[This GHA run](https://github.com/NUbots/NUbots/pull/728/checks?check_run_id=3780850190#step:8:35) isolates the problem reasonably well. Single threaded, [after all the code in the test case](https://github.com/NUbots/NUbots/pull/728/files#diff-ba446b102ba87bc70b1dacd1ec0c8d21b1581e6bd1a5cf509ba45fbfeef14f03R78), we get inside the destructor for FW, then promptly segfault. [Somewhere in that first `for` loop](https://github.com/NUbots/NUbots/pull/728/files#diff-0e68052f9d92d448f711689136865870e93be8a480a5aaf25e826292fde4a430R296-R301), it's having the problem.

@CMurtagh-LGTM do you have experience with FileWatcher? Any ideas?

Also, if anyone can replicate this error, that would make it much much easier to debug - adding print statements to send to GHA to execute after building for 20 mins is less than ideal lol.

This currently blocks #728 and transitively #727, since I can't merge the test API in good faith when the only demo segfaults lmao

JosephusPaye, 2021-10-16:
Oooh so it was in a destructor

KipHamiltons, 2021-10-16:
Yessir. I have no idea why tho :(

CMurtagh-LGTM, 2022-07-08:
```
Start of filewatcher destructor
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TestmotionKinematicsConfiguration is a Catch v2.11.1 host application.
Run with -? for options
-------------------------------------------------------------------------------
Testing the Kinematics Configuration module
-------------------------------------------------------------------------------
/__w/NUbots/NUbots/module/motion/KinematicsConfiguration/tests/KinematicsConfigurationTest.cpp:55
...............................................................................
/__w/NUbots/NUbots/module/motion/KinematicsConfiguration/tests/KinematicsConfigurationTest.cpp:55: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal
===============================================================================
test cases: 1 | 1 failed
assertions: 2 | 1 passed | 1 failed
```
The runs you had linked have not been retained, here is the output of a newer run.

CMurtagh-LGTM, 2022-07-08:
@KipHamiltons I'm having a hard time reproducing this on my local machine: I've tried running with a clean configure but that didn't work, I've made sure my docker image is latest, I've made sure that I'm using the same cmake configuration but I cannot get the tests to fail.

KipHamiltons, 2022-07-08:
@CMurtagh-LGTM i could never reproduce this on my machine. The only place it would happen is on the GitHub cloud runners running with parallel tests (I think because there were multiple file watchers? The issue happened during a C library call and I went deep with print statements but couldn't put it together - it's quite slow, waiting for the GitHub runners too)

closed 2022-07-09 by CMurtagh-LGTM