NUbots/NUbots #742: Localisation Potential Fix

tayloryoung6396, 2021-10-06:
In _RobotParticleLocalisation.cpp_ we currently take the _Eigen::Vector3d state(filter.getMean());_ as our state.. this seems bad @Bidski and myself seem to think this should either be just the most confident particle, OR an average of the top few. This could be the reason why our localisaiton doesn't converge to the correct side of the field

tayloryoung6396, 2021-10-06:
So then if the filter is designed for a single hypothesis.. we should try seeding it with the correct on (manually) to check that this work. Then if that worked we can use some last N measurements of things to seed a better initial hypothesis (from our few we know) and use that as our filter seed

ysims, 2026-09-20:
Out of date

closed 2026-09-20 by ysims