NUbots/NUbots #780: Use Pacman for Package Management

Mrwerdo, 2022-01-19:


Mrwerdo, 2022-01-19:
Another concern is cross-compilation. Currently packages are being compiled for both a generic architecture and specifically `nuc7i7bnh`, among others. All of the dependencies still need to be built with the cross-compilation toolchain.

KipHamiltons, 2022-01-20:
If we can't build the container, that's definitely something worth fixing. I personally don't mind the first option you present.

Also, could you please elaborate on how the second option would leverage patches from the arch community? I don't quite understand how that would be the case (unless we didn't pin the dependencies, which is not preferred), since we can already use their patches by getting them manually.

CC @NUbots/devops

Mrwerdo, 2022-01-20:
AUR packages are maintained against the current version of Arch Linux, right? So that means that anyone wanting to use packages that we have version locked will need to build themselves (with the latest version). If they find the build fails, they might be bothered to fix it, and they might be bothered to share it with us, at which point we benefit because someone else has figured out the problem and solution before we have looked at it.

As for version locking, could we use environment variables to override the version of packages? Obviously the PKGBUILD files would need to be modified to support it.

Having said that, there'a two parts to problem:
- Arch Linux core packages being updated.
- Upstream packages changing versions.

Maybe it's not the best approach because it could be awkward maintaining PKGBUILD script at both ends of the spectrum (lock package + locked system, and latest package + latest system). Idk. That's why I'm discussing it here. 

I feel that it took a bit of work figuring out how to get the various dependencies to build together and that it would be nice if the community could benefit from it somehow.

Bidski, 2022-06-11:
If I am not mistaken `makepkg` hates being run as `root`, in fact it refuses to run if you are `root`. We also happen to do all of our docker building as `root`, so you would need to re-architect how the docker file works. Not particular difficult to do, but something that would need doing.

Using `makepkg` won't speed up the build time, you still have to build all of the packages yourself, and keeping the tarballs around so you can copy them out later will bloat the size of the docker image.

You would need to ensure that the packages built for the `nuc7i7bnh` image have the appropriate optimisations applied to them. Once again, not particularly difficult to do, but I think it would have to be done differently to how we currently do it.

AUR packages are community maintained. You have no guarantees that they are maintained against any particular version of Arch Linux, many of them are probably regularly out of date, either with the latest version of Arch Linux or with the latest version of the package itself. I am also not aware of any way of installing a particular version of a package from AUR, short of downloading the PKGBUILD and manually editting it. Also, considering the extreme end of the scale, you also have no guarantees that the PKGBUILD scripts are safe to run (as I said, this is extreme, but a PKGBUILD script is just a bash script and can run any of the commands that your user can run).

If the Dockerfile is working (and being used) as intended, the addition of new dependencies should be a minor rebuild as the new dependencies should be added near the end of the file.

Ultimately, I dont see a lot of benefit to moving to a `makepkg`-based build system. I considered this option when @TrentHouliston and I were implementing all of this Dockerfile stuff originally and for most of the reasons I have outlined we rejected it. We leverage what we can from the community when we can (its not uncommon to use a PKGBUILD file as inspiration for one of our custom scripts), but the majority of the things we are building are well enough "behaved" that custom scripts and patches are not really needed most of the time.