NUbots/NUbots #868: Turbojpeg on the robot has a memory leak

ysims, 2022-07-01:
Should be fixed. Vaapi works. Turbojpeg works for webots.

Bidski, 2022-07-01:
Is it `turbojpeg` that has the memory leak, or the image compressor that uses `turbojpeg` ?

TrentHouliston, 2022-07-01:
My guess is it would be our usage, turbojpeg is a very mature library

CMurtagh-LGTM, 2022-07-05:
@ysims commented that vaapi also leaks when run in the visual mesh role.

ysims, 2022-07-05:
and sometimes seg faults. Behaviour is inconsistent though, for example `cameratest` doesn't seem to have the issue but `visualmesh` does.

Bidski, 2022-07-05:
more information is needed, how do you know the issue is with turbojpeg? how do you know the issue is with the image compressor? does it happen in a debug build? are there any stack traces? have you run it with ASAN?

CMurtagh-LGTM, 2022-07-05:
I'm looking into it now.

TrentHouliston, 2022-07-05:
vaapi uses the GPU with OpenCL just like the VisualMesh does, so there is a chance you're just running out of something on the GPU?

CMurtagh-LGTM, 2022-07-05:
For future reference for myself.
```
VisualMesh (╯°□°）╯︵ ┻━┻  std::runtime_error Segmentation Fault
	/home/nubots/NUbots/module/support/SignalCatcher/src/SignalCatcher.cpp:55 module::support::sigsegv(int)
	/home/nubots/NUbots/module/vision/VisualMesh/src/visualmesh/VisualMeshRunner.cpp:91 module::vision::visualmesh::generate_runner::runner<visualmesh::model::Ring6, visualmesh::engine::opencl::Engine, visualmesh::geometry::Sphere<double> >(module::vision::visualmesh::VisualMeshModelConfig const&, visualmesh::geometry::Sphere<double> const&)::{lambda(message::input::Image const&, Eigen::Transform<float, 3, 2, 0> const&)#1}::operator()(message::input::Image const&, Eigen::Transform<float, 3, 2, 0> const&) const
	/usr/include/c++/11.1.0/bits/std_function.h:560 std::function<module::vision::visualmesh::VisualMeshResults (message::input::Image const&, Eigen::Transform<float, 3, 2, 0> const&)>::operator()(message::input::Image const&, Eigen::Transform<float, 3, 2, 0> const&) const
	/home/nubots/NUbots/module/vision/VisualMesh/src/visualmesh/VisualMeshRunner.cpp:208 module::vision::visualmesh::VisualMeshRunner::operator()(message::input::Image const&, Eigen::Transform<float, 3, 2, 0> const&)
	/home/nubots/NUbots/module/vision/VisualMesh/src/VisualMesh.cpp:61 operator()
	/usr/local/include/nuclear_bits/util/apply.hpp:48 apply<const module::vision::VisualMesh::VisualMesh(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)>&, 0, std::shared_ptr<const message::input::Image> >
	/usr/local/include/nuclear_bits/util/apply.hpp:55 apply_relevant<const module::vision::VisualMesh::VisualMesh(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)>&, std::shared_ptr<const message::input::Image> >
	/usr/local/include/nuclear_bits/util/CallbackGenerator.hpp:110 operator()
	/usr/include/c++/11.1.0/bits/invoke.h:61 __invoke_impl<std::unique_ptr<NUClear::threading::ReactionTask>, NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<message::input::Image> >, module::vision::VisualMesh::VisualMesh(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
	/usr/include/c++/11.1.0/bits/invoke.h:116 __invoke_r<std::unique_ptr<NUClear::threading::ReactionTask>, NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<message::input::Image> >, module::vision::VisualMesh::VisualMesh(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
	/usr/include/c++/11.1.0/bits/std_function.h:292 _M_invoke
	/usr/src/debug/gcc/libstdc++-v3/src/c++11/thread.cc:82 execute_native_thread_routine
^C
Shutdown Command Sent
ImageCompressor - Compress Image (╯°□°）╯︵ ┻━┻  std::runtime_error Segmentation Fault
	/home/nubots/NUbots/module/support/SignalCatcher/src/SignalCatcher.cpp:55 module::support::sigsegv(int)
	/home/nubots/NUbots/module/output/ImageCompressor/src/compressor/vaapi/cl/mosaic_to_surface.cpp:90 module::output::compressor::vaapi::cl::mosaic_to_surface(module::output::compressor::vaapi::CompressionContext::OpenCLContext const&, module::output::compressor::vaapi::cl::opencl_wrapper<_cl_command_queue*> const&, module::output::compressor::vaapi::cl::opencl_wrapper<_cl_kernel*> const&, module::output::compressor::vaapi::cl::opencl_wrapper<_cl_mem*> const&, std::vector<unsigned char, std::allocator<unsigned char> > const&, unsigned int const&, unsigned int const&, _cl_mem* const&)
	/home/nubots/NUbots/module/output/ImageCompressor/src/compressor/vaapi/Compressor.cpp:120 module::output::compressor::vaapi::Compressor::compress(std::vector<unsigned char, std::allocator<unsigned char> > const&)
	/home/nubots/NUbots/module/output/ImageCompressor/src/ImageCompressor.cpp:121 operator()
	/usr/local/include/nuclear_bits/util/apply.hpp:48 apply<const module::output::ImageCompressor::ImageCompressor(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)>&, 0, std::shared_ptr<const message::input::Image> >
	/usr/local/include/nuclear_bits/util/apply.hpp:55 apply_relevant<const module::output::ImageCompressor::ImageCompressor(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)>&, std::shared_ptr<const message::input::Image> >
	/usr/local/include/nuclear_bits/util/CallbackGenerator.hpp:110 operator()
	/usr/include/c++/11.1.0/bits/invoke.h:61 __invoke_impl<std::unique_ptr<NUClear::threading::ReactionTask>, NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<message::input::Image> >, module::output::ImageCompressor::ImageCompressor(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
[... cut ...] ompressor(std::unique_ptr<NUClear::Environment>)::<lambda(const message::input::Image&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
	/usr/include/c++/11.1.0/bits/std_function.h:292 _M_invoke
	/usr/src/debug/gcc/libstdc++-v3/src/c++11/thread.cc:82 execute_native_thread_routine
^C
Shutdown Command Sent
nubots@nugus3 ~ ❯❯❯
```

CMurtagh-LGTM, 2022-07-07:
https://github.com/intel/compute-runtime/pull/550 fixes the ASan output.

CMurtagh-LGTM, 2022-08-05:
```
==498==ERROR: AddressSanitizer: heap-use-after-free on address 0x6060004eefc0 at pc 0x5613e274a85f bp 0x7f2dada93dd0 sp 0x7f2dada93dc0
```
So its a use after free we are looking fore :thinking:

CMurtagh-LGTM, 2022-08-05:
When running without the signal catcher we get
```
==1583==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fb0f1236938 bp 0x618000051480 sp 0x7fb0a61f7920 T12)
```
So perhaps its a nullptr we're dereferencing.

ysims, 2024-01-20:
This is no longer a problem since updating libraries.

closed 2024-01-20 by ysims