NUbots/NUbots #877: Move github pages to doxygen.nubots.net

CMurtagh-LGTM, 2022-07-05:
Currently we host our doxygen output on nubots.github.io/NUbots Trent suggested putting it on nubots.net as this is were we have NUbook.
https://docs.github.com/en/pages/configuring-a-custom-domain-for-your-github-pages-site/managing-a-custom-domain-for-your-github-pages-site

JosephusPaye, 2022-07-05:
We currently have the nubots.net domain managed in Netlify. Let me see about adding the subdomain.

LachlanCourt, 2022-07-05:
I would suggest the url just be `docs.nubots.net` just so that it's one less thing to change if we decide to move away from doxygen in the future

JosephusPaye, 2022-07-05:
Good point. Other options could be `codedocs.` or `api.` perhaps? Thinking of someone getting confused with `docs.` not being going to NUbook as NUbook is technically also "docs".

JosephusPaye, 2022-07-06:
It's done: https://codedocs.nubots.net/