NUbots/NUbots #939: Director test segfaults

ysims, 2022-09-03:
CI won't pass if it's rerunning tests. 

https://github.com/NUbots/NUbots/blob/643912923291820f220142914a063fab187d4dbe/module/extension/Director/tests/triggered_provider.cpp#L58
It's in the triggered_provider test. When the reactors are commented out except for

- `Startup` with just `Step<1>` 
- `on<Trigger<Step<1>>,Priority::LOW>` with just the second line (emit OtherData(0))
- `on<Provide<SimpleTask>, Trigger<OtherData>>` is there, but nothing in it except an INFO to see if it runs (it does not).

The Provide<SimpleTask> reactor has to be there and then it only happens if emit OtherData is there too. 

ASAN gives

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==26==ERROR: AddressSanitizer: SEGV on unknown address 0x03e80000001a (pc 0x7fce9a17f550 bp 0x7fce930fbbb0 sp 0x7fce930fb5a0 T33)
==26==The signal is caused by a READ memory access.
    #0 0x7fce9a17f550 in module::extension::Director::_get_task_data(unsigned long const&) (bin/lib/libextensionDirector.so+0x163550)
    #1 0x55ac5f14a873 in std::_Function_handler<std::pair<int, std::function<std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > (std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> >&&)> > (NUClear::threading::Reaction&), NUClear::util::CallbackGenerator<NUClear::dsl::Parse<extension::behaviour::Provide<(anonymous namespace)::SimpleTask>, NUClear::dsl::word::Trigger<(anonymous namespace)::OtherData> >, (anonymous namespace)::TestReactor::TestReactor(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda((anonymous namespace)::OtherData const&)#1}> >::_M_invoke(std::_Any_data const&, NUClear::threading::Reaction&) (/home/nubots/build/module/extension/Director/TestextensionDirector+0x5cf873)
    #2 0x7fce9a1835c1 in NUClear::threading::Reaction::get_task() (bin/lib/libextensionDirector.so+0x1675c1)
    #3 0x55ac5f161e3c in (anonymous namespace)::TestReactor::TestReactor(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda()#2}::operator()() const (/home/nubots/build/module/extension/Director/TestextensionDirector+0x5e6e3c)
    #4 0x55ac5f162ddc in std::_Function_handler<std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > (std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> >&&), NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<TestBase<(anonymous namespace)::TestReactor, 1000>::Step<1> >, NUClear::dsl::word::Priority::LOW>, (anonymous namespace)::TestReactor::Parse(std::unique_ptr<NUClear::Environment, std::default_delete<NUClear::Environment> >)::{lambda()#2}>::operator()(NUClear::threading::Reaction&)::{lambda(std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> >&&)#1}>::_M_invoke(std::_Any_data const&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> >&&) (/home/nubots/build/module/extension/Director/TestextensionDirector+0x5e7ddc)
    #5 0x7fce9a183d1b in NUClear::threading::ReactionTask::run(std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> >&&) (bin/lib/libextensionDirector.so+0x167d1b)
    #6 0x7fce9a182350 in std::_Function_handler<void (), NUClear::threading::make_thread_pool_task(NUClear::threading::TaskScheduler&)::{lambda()#1}>::_M_invoke(std::_Any_data const&) (bin/lib/libextensionDirector.so+0x166350)
    #7 0x7fce97b8f3c3 in execute_native_thread_routine /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82
    #8 0x7fce97cdb258 in start_thread (/usr/lib/libpthread.so.0+0x9258)
    #9 0x7fce9788f5e2 in __GI___clone (/usr/lib/libc.so.6+0xfe5e2)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (bin/lib/libextensionDirector.so+0x163550) in module::extension::Director::_get_task_data(unsigned long const&)
Thread T33 created by T0 here:
    #0 0x7fce9a24cfa7 in __interceptor_pthread_create /build/gcc/src/gcc/libsanitizer/asan/asan_interceptors.cpp:216
    #1 0x7fce97b8f6aa in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu/bits/gthr-default.h:663

==26==ABORTING
```

TrentHouliston, 2022-09-03:
Is it still broken in #915 ?

ysims, 2022-09-03:
No, seems to be fine (and the test runs successfully) in #915.

closed 2022-11-01 by CMurtagh-LGTM