NUbots/NUbots #962: Controller module sometimes segfaults

ysims, 2022-10-08:
Some output from @PiiDish 

```
    /usr/include/c++/11.1.0/bits/std_function.h:292 _M_invoke
    /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82 execute_native_thread_routine
Controller - Command Filter (╯°□°）╯︵ ┻━┻  std::runtime_error Segmentation Fault
    /home/nubots/NUbots/module/support/SignalCatcher/src/SignalCatcher.cpp:55 module::support::sigsegv(int)
    /home/nubots/NUbots/module/behaviour/Controller/src/Controller.cpp:140 operator()
    /usr/local/include/nuclear_bits/util/apply.hpp:48 apply<const module::behaviour::Controller::Controller(std::unique_ptr<NUClear::Environment>)::<lambda(const message::behaviour::ServoCommands&)>&, 0, std::shared_ptr<const message::behaviour::ServoCommands> >
    /usr/local/include/nuclear_bits/util/apply.hpp:55 apply_relevant<const module::behaviour::Controller::Controller(std::unique_ptr<NUClear::Environment>)::<lambda(const message::behaviour::ServoCommands&)>&, std::shared_ptr<const message::behaviour::ServoCommands> >
    /usr/local/include/nuclear_bits/util/CallbackGenerator.hpp:110 operator()
    /usr/include/c++/11.1.0/bits/invoke.h:61 __invoke_impl<std::unique_ptr<NUClear::threading::ReactionTask>, NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<message::behaviour::ServoCommands>, NUClear::dsl::word::Sync<module::behaviour::Controller> >, module::behaviour::Controller::Controller(std::unique_ptr<NUClear::Environment>)::<lambda(const message::behaviour::ServoCommands&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
    /usr/include/c++/11.1.0/bits/invoke.h:116 __invoke_r<std::unique_ptr<NUClear::threading::ReactionTask>, NUClear::util::CallbackGenerator<NUClear::dsl::Parse<NUClear::dsl::word::Trigger<message::behaviour::ServoCommands>, NUClear::dsl::word::Sync<module::behaviour::Controller> >, module::behaviour::Controller::Controller(std::unique_ptr<NUClear::Environment>)::<lambda(const message::behaviour::ServoCommands&)> >::operator()(NUClear::threading::Reaction&)::<lambda(std::unique_ptr<NUClear::threading::ReactionTask>&&)>&, std::unique_ptr<NUClear::threading::ReactionTask, std::default_delete<NUClear::threading::ReactionTask> > >
    /usr/include/c++/11.1.0/bits/std_function.h:292 _M_invoke
    /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82 execute_native_thread_routine
```

If someone who has seen it can give some more info that would be good, particularly on how to reproduce the seg fault. Is it only in Webots?
@PiiDish @Crafty15

ysims, 2022-12-29:
@PiiDish @Crafty15 is this still a thing or was it just user error?

Crafty15, 2022-12-29:
> @PiiDish @Crafty15 is this still a thing or was it just user error?

I can't remember if we came to a conclusion as to the cause. But it hasn't popped up for a long time.

PiiDish, 2022-12-29:
I haven't encountered it again. As far as I know, it's been righted

ysims, 2022-12-30:
Unlikely to be an issue - and Controller will be removed soon in favour of The Director.

closed 2022-12-30 by ysims