NUbots/NUbots #975: Port the current behaviour system to the Director

TrentHouliston, 2022-10-29:
Porting everything to the director will require transforming/creating modules to the director.
This issue exists to track what still remains

### Motion

- [X] Servo #973
- [X] Servo Groups #973
- [X] IK #973
- [ ] Servo Sequences
- [ ] Scripts

### Skills

- [ ] Getup
- [ ] Falling Relax
- [ ] Walk
- [ ] LookAt
- [ ] Kick

### Planners

- [ ] Walk path planner
- [ ] Look Planner
- [ ] Kick Planner

### Strategy

- [ ] WalkToBall
- [ ] KickTowardsGoal

### Purpose

- [ ] Striker
- [ ] Goalie
- [ ] RoboCup

ysims, 2022-12-15:
[actuation, motion, skills, planning, strategy, purpose]

ysims, 2023-02-11:
Reminder to self about the watcher seg fault when the watchers are equal (remove_watcher_double with no priority set).
Also the memory problems in director tests, unrelated.

ysims, 2023-06-17:
Closing this since "porting behaviour system to Director" is done and bugs are moved to #1125.

closed 2023-06-17 by ysims