NUbots/NUbots #1033: Robot Localisation

ysims, 2023-03-10:
Tasks for localisation

- [ ] Base algorithm (occupany grid method) #1030 
- [ ] Update SensorFilter with a more reliable filter for pitch, roll and z height
- [ ] Update field line detector module so it doesn't need to call the cluster method
- [ ] Reset functionality, eg for penalty
- [ ] Add goal measurements
- [ ] Use odometry in time update (move walk command to SensorFilter?)
- [ ] Put particle filter into a generic utility to use in RobotLocalisation
- [ ] Fix NUsight localisation view

ysims, 2023-03-11:
@JosephusPaye the NUsight localisation view might be up your alley if you feel like working on it? The problem is that the localisation view takes odometry rather than localisation info.

JosephusPaye, 2023-03-11:
Yeah I can have a look. I'll be on campus on Tuesday and will stop by the lab.